Robo3D/User-Manuals #3: Subdomain and DNS need to be setup

AllenMcAfee, 2016-12-11:
Current site is located at: http://support-site.readthedocs.io/

The site is generated by Readthedocs.org.

desired subdomain address can be: docs.robo3d.com  or    manual.robo3d.com.  Open to other ideas.

AllenMcAfee, 2016-12-11:
Language support question: I provided the base url for the supporting docs site, that is the address before the language.  The full address is like this: ```http://support-site.readthedocs.io/en/latest/?badge=latest```.  

In the final deliverable, we want to host multiple languages.  Readthedocs can provide language support in the method I showed above.  We want to make multilanguage browsing seamless by auto-updating all the links on the new help.robo3d.com landing page to correctly link to preselected language versions of both this docs site and the forum.  

So the question Is: Should subdomains be used for this?

CobyKabili, 2017-05-17:
@bkabili will do this when he gets access to the hosting info when @braydonmoreno gets back on thursday or Friday.

AllenMcAfee, 2017-05-17:
Credentials for godaddy are:

morenob
CreateGreat100!!!!

Brictone, 2017-05-17:
For some reason on still on this. Please change the godaddy password and
make sure me or anyone else that should not be is not on this guthub. Also
sharing passwords over messages and email is never a good idea.

bkabili, 2017-05-18:
This is complete.

closed 2017-05-18 by CobyKabili