Robo3D/User-Manuals #7: Need to add a folder

dzimma, 2016-12-16:
I want to add new sections in the manual that line up correctly with what's in the Google doc.

dzimma, 2016-12-16:
@AllenMcAfee 

Let's make sure we have the correct sections in the manual based on what's in the Google doc.

dzimma, 2016-12-16:
@AllenMcAfee 

For last step of Loading Filament under Getting Started I need the following shown as a bullet list:

Heating up the printer 
Ensuring you’ve cut off the tip of the filament 
Having the filament feed through until you see it coming from the nozzle 
Pressing NEXT, and 
Pressing FINISHED

dzimma, 2016-12-16:
@AllenMcAfee 

For last step of Calibration and setting Z offset under Getting Started I need the following shown as a bullet list:

Waiting for the printer to calibrate its home position
Pressing up and down on the [insert descriptor] buttons while sliding a piece of paper between the nozzle and the print bed until you feel some resistance against the paper
Pressing FINISHED

dzimma, 2016-12-16:
@AllenMcAfee 

For Step 4 and beyond for Using Hotspot Mode under Getting Started, we'll need the language to complete it. Right now, what Coby provided is insufficient (we have the images he provided copied into the Google Doc, but the text on them is hard to read).

AllenMcAfee, 2016-12-28:
@dzimma I've fixed the bullet list requests, but can you list out the correct sections here?  The google doc is hard to sift through.  I will try to get the images working then work on the hotspot language.

dzimma, 2016-12-28:
@AllenMcAfee 

I have to push back some here to get some clarity on the email I sent out to you on 12/20 via gmail:

"Hi Allen,

  I've loaded most of the content via Github.

  However, I noticed there is some overlap with what we want in the Getting Started section and what's currently in the Operations section (especially in regards to setting the Z offset, removing a print, setting up wifi, and downloading the Robo app... which I've included up front on the Robo Support Documentation page).

  Also hesitant to change what's been inputted in the Software section, since it's not quite aligned with what I've worked out in the Google doc.

  https://docs.google.com/document/d/1pl07DrfMT5aVIou51A5mKLG6RA6_EMNlOrOTFJpcIyw/edit?ts=5819ec83# 

  Will also need you to add a Troubleshooting section after Software, so all of that copy can be ported in from the Google doc.

  Hope this makes sense. Let me know if we need to hop on a call."

closed 2017-04-04 by CobyKabili