Robo3D/User-Manuals #23: Move the Read The Docs black bar back under menu list

AllenMcAfee, 2017-04-20:
@bkabili @CobyKabili Ben had moved the black bar I'm showing in the attached picture out from under the menu list, but on smaller screens is blocks out the links for the bottom of the lists as they expand.  

Can we move this back inline with the menu?

[image]

bkabili, 2017-04-20:
@AllenMcAfee I added a larger margin to the bottom of the menu so that you can always scroll to the bottom of the menu even if you are on a small screen and the black bar is over the menu. You should be able to get to all parts of the menu no matter what. 
[image]
In the out of the box Read the Docs styling, the bar covers the menu as well and you can scroll to the bottom similar to how we have it set now. Would you rather have it a different way?
[image]

bkabili, 2017-06-12:
@AllenMcAfee @CobyKabili can we close this or do we still want the bar moved?

AllenMcAfee, 2017-06-12:
@bkabili It is still problematic in it's current state.  Check the photo below.
[image]

The bar still overs the text if the menu is long enough.  If we can move this directly under the menu, and shrink the menu's bottom margin to fit it, then I think we're in business.

closed 2017-10-28 by CobyKabili