Robo3D/User-Manuals #30: Create webpage robo3d.com/calibration

AllenMcAfee, 2017-05-11:
The Zoffset tool has the weblink `robo3d.com/calibration` at the bottom of the card.  We need to make this page before R2s arrive with customers.

CobyKabili, 2017-05-18:
@AllenMcAfee do we have a specific date or at least an estimated date an R2 will get to a customer?

AllenMcAfee, 2017-05-18:
@jerryrobo3d any thoughts on the date R2 customers will start receiving their printers?

CobyKabili, 2017-05-25:
@bkabili for now can you redirect robo3d.com/calibration to http://docs.robo3d.com/en/latest/operation/index.html#setting-the-z-offset

bkabili, 2017-05-25:
robo3d.com/calibration is now directing to http://docs.robo3d.com/en/latest/operation/index.html#setting-the-z-offset

bkabili, 2017-06-12:
@CobyKabili this url is no longer valid. 
I can change the redirect. Should I use 
http://docs.robo3d.com/en/latest/C2/operation/index.html#setting-the-z-offset
or
http://docs.robo3d.com/en/latest/R2/operation/index.html#setting-the-z-offset

CobyKabili, 2017-06-12:
Use http://docs.robo3d.com/en/latest/C2/operation/index.html#setting-the-z-offset 
for now until we get the R2 support page up to speed