Robo3D/User-Manuals #37: Headers needed for R2

CobyKabili, 2017-06-02:
@bkabili I need these headers to be made before we push the doc live

- [ ] Getting Started

- [ ] Operation

- [ ] Maintenance

- [ ] Troubleshooting

CobyKabili, 2017-06-02:
Here are the renders https://www.dropbox.com/home/R2%20renders

bkabili, 2017-06-02:
[image]
[image]
[image]
[image]
[image]
[image]

CobyKabili, 2017-06-02:
Thanks @bkabili the Robo R2 maintenance is a picture of the C2 though...

bkabili, 2017-06-02:
@CobyKabili ok, lets just use this then.
[image]

CobyKabili, 2017-06-02:
Awesome thanks!

closed 2017-06-02 by CobyKabili