Robo3D/User-Manuals #38: R2- starting the webcam

CobyKabili, 2017-06-02:
User is confused as to how to start the webcam. Didnt know it ships with it off - need to show how to turn it on.

CobyKabili, 2017-06-20:
@Ximidar is this in the latest release? Is there any special instructions beside toggling it on and off from the screen?

Ximidar, 2017-06-20:
@CobyKabili There is only a toggle on the LCD screen. I believed it was pushed to the users, but we will need @victorevector to verify. There is no other information on turning the webcam on and off

CobyKabili, 2017-06-21:
@Ximidar @victorevector what is the function that this button does? (I assume its just starting the webcam) Does the user need to refresh the browser once it is pressed?

victorevector, 2017-06-21:
@CobyKabili there are currently 3 official ways to turn the webcam on and off:

1. RoboLCD: Utilities > Options > Webcam

2. Web Dashboard: Top Right Menu > System > Start Webcam

3. iOS (BETA only): Webcam > toggle button

closed 2017-06-21 by CobyKabili