Robo3D/User-Manuals #39: Make a 'latest' page that has both C2 AND R2 information

CobyKabili, 2017-06-02:
Right now, the latest page is the first thing that comes up, but there is only info on the C2. We need to change this to be both products.

http://docs.robo3d.com/en/latest/index.html

@bkabili can you make a header that show both c2 and r2 and maybe the app and a comp with cura on it? Anything that looks good in your opinion to be the header for all documentation

bkabili, 2017-06-02:
@CobyKabili How about this?
[image]

CobyKabili, 2017-06-02:
@bkabili Can it say docs instead of support

bkabili, 2017-06-05:
@CobyKabili 
[image]

CobyKabili, 2017-06-05:
perfect

closed 2017-06-05 by CobyKabili