Robo3D/User-Manuals #56: shut down procedure

CobyKabili, 2017-06-13:


mnacinopa, 2017-09-08:
Added to Helpdesk articles [Correct Shutdown Sequence](https://help.robo3d.com/hc/en-us/articles/115001490471-Correct-Shutdown-Sequence)

closed 2017-09-08 by mnacinopa

peries, 2017-09-08:
@mnacinopa a few edits:
Make sure to follow the proper **shut down** procedure to help keep your machine running smoothly. **Do not turn off the back power switch before running the on-screen shut down  sequence.**

**On the printer's touchscreen, navigate** to Options >> System >> Shut Down
Follow **the** on-screen instructions and wait **until** the screen turns black.
Once the screen turns black, shut off the printer using the **power** switch located on the back of **the printer.**

mnacinopa, 2017-09-08:
@peries  Got it. Updated