Robo3D/User-Manuals #63: Cura windows- dll error on hp (w/ fix)

CobyKabili, 2017-07-10:
Download http://ftp.hp.com/pub/softpaq/sp78001-78500/sp78033.exe and extract the files.
Open windows Control Panel, open Programs and Features, right click the entry for HP 3D DriveGuard and select Uninstall.
(If it is not in the list, just run the setup and select Remove if that option is available)
When this has completed, restart the computer
Run the setup again, if you get an INI error during setup, it doesn't matter, the necessary files have been deployed,  just close it.
Cura for Robo should now run properly.

AllenMcAfee, 2017-07-10:
@CobyKabili can you provide some info on how you got to this?

CobyKabili, 2017-07-10:
@AllenMcAfee Dan from CS said it happens only on HP computers. This was a reference just to put into the troubleshooting  to make it easier for the CS guys, but I do not know the root cause, only that it sometimes comes up as a customer service email. It happened one time on steves microsoft surface but that was when we deployed 2.3 for the very first time. 2.5 seems to only have issues on HP computers.

AllenMcAfee, 2017-07-10:
@CobyKabili thanks!  Has this been tested?

closed 2017-07-17 by CobyKabili