Robo3D/User-Manuals #73: Mobile view shows `Stash changes` header in menu

AllenMcAfee, 2017-07-26:
[image]

CobyKabili, 2017-07-26:
@bkabili this is weird ive never seen that. Maybe because there is more content then we are showing? I dont know how to fix this

bkabili, 2017-07-26:
@AllenMcAfee @CobyKabili I noticed when you refresh the page on a desktop browser you can see this same message flash quickly and then go away. Ill look in to this.

bkabili, 2017-07-28:
@AllenMcAfee this is from a git merge conflict:
https://wincent.com/wiki/Git_merge_conflict_cheatsheet

Coby said you would know best how to resolve something like this.

AllenMcAfee, 2017-07-28:
This can be a GitHub merge error. There is really no way to reverse this
that I know of, you'll need to scrub these bugged lines by hand.

CobyKabili, 2017-10-05:
this is outdated for the moment. Can be reopened if we decide to use it for other projects

closed 2017-10-05 by CobyKabili