Robo3D/User-Manuals #88: R2 Maintenance Guides

ROBOLabs, 2017-08-23:
@CobyKabili Here are the Gantry Removal, Front Panel Removal, and LCD Replacement Guides. More guides to come soon.
[R2 Front Panel Removal Guide.pdf](https://github.com/Robo3D/support.robo3d.com/files/1246779/R2.Front.Panel.Removal.Guide.pdf)
[Robo 3D R2 LCD Screen Replacement Guide.pdf](https://github.com/Robo3D/support.robo3d.com/files/1246780/Robo.3D.R2.LCD.Screen.Replacement.Guide.pdf)
[R2 Gantry Removal Guide.pdf](https://github.com/Robo3D/support.robo3d.com/files/1246781/R2.Gantry.Removal.Guide.pdf)

mnacinopa, 2017-08-23:
[Front Panel Removal](https://docs.google.com/document/d/1bNjgpHLnlV-nBt2RHenPyoMlTDSq3BNYZuP1_3nop5s/edit?ts=599cd477)

@ROBOLabs  Can you post the rest of the editable docs? We need to be able to parse the images from the document. So like the original word doc, or the google doc link. Thanks

AllenMcAfee, 2017-08-23:
@ROBOLabs can we also make Google Slides our default style for these tutorials?  The landscape view is much nicer and we have a template ready made for Robo.

ROBOLabs, 2017-08-23:
[LCD Replacement Guide](https://docs.google.com/a/robo3d.com/document/d/1ci8q527K_u_o4PlOvhNRXqdOw1pbAyriQBRV8H4-G-4/edit?usp=sharing)

[Gantry Removal Guide](https://docs.google.com/a/robo3d.com/document/d/1B4C_Hxn6ZvLPZgTgBLzZ5toGPrLozUEurjw3Fsm48lY/edit?usp=sharing)

@AllenMcAfee Noted. Will use Slides with Robo template moving forward.

mnacinopa, 2017-08-28:
First draft of the [Assembly Docs](http://support-site.readthedocs.io/en/dev/R2/assembly/index.html) for both the Gantry and Front Panel Removal.

mnacinopa, 2017-08-28:
Added LCD Replacement.

closed 2017-11-07 by CobyKabili