Robo3D/User-Manuals #90: Loosening and tightening pulleys on R2 and C2

CobyKabili, 2017-09-13:
Tutorial showing how to loosen and tighten pulleys on R2 and C2 for fixing spacing (maybe the creeking noise) - dont forget to use the rod holders for keeping the gantry alligned. May need to put these up on thingiverse for users to print before they start the tutorial.

mnacinopa, 2017-09-27:
[Gantry Alignment](https://help.robo3d.com/hc/en-us/articles/115001792512). Here's a rough pass at a gantry aligning tutorial.

mnacinopa, 2017-09-28:
@CobyKabili  Revised it! Let me know if you want it done any differently.

CobyKabili, 2017-10-05:
still not quit right. Ill let you know, but its almost there.

CobyKabili, 2017-10-05:
@mnacinopa all 4 pulleys of one axis have to be loosened. Also, if there can be a picture highlighting which pulleys to loosen it would be nice.

CobyKabili, 2017-10-31:
@peries can you review these and make sure they make sense to you

mnacinopa, 2017-11-01:
@CobyKabili  I added the picture highlighting the pulleys into the R2 article

peries, 2017-11-01:
@CobyKabili made a couple tiny edits but it looks good to me

CobyKabili, 2017-11-01:
@mnacinopa can you copy this to the C2 page and just put a note that it is the same process. Then you can close out this issue.

mnacinopa, 2017-11-01:
@CobyKabili @peries  

[Its liveee. ](https://help.robo3d.com/hc/en-us/articles/115002529251?flash_digest=6e45df654650a65417329aaba7b78704720b4caa)

closed 2017-11-01 by CobyKabili