Robo3D/User-Manuals #91: Troubleshooting - R1 - check the wire for bad extruder motor and how to replace it

CobyKabili, 2017-09-13:
check stepper driver as well

mnacinopa, 2017-09-15:
@CobyKabili  Can you show me how to troubleshoot this at some point?

mnacinopa, 2017-10-17:
@CobyKabili  I Created the troubleshooting portion of the issue. Published it as a draft. Next I'll work on actual replacement of the parts.

mnacinopa, 2017-10-18:
@CobyKabili @peries   Finished updating it for replacing the stepper driver. Please review/edit grammar as you see fit and go ahead and publish it. 
[Extruder Motor Not Working](https://help.robo3d.com/hc/en-us/articles/115002197612)

CobyKabili, 2017-10-18:
@mnacinopa I don't see steps 5-8 for some reason...

mnacinopa, 2017-10-18:
@CobyKabili  Weird. It deleted them. I remade the steps.

CobyKabili, 2017-10-19:
@mnacinopa this looks great! I edited the last version and pushed it live. Good work, this was a big one we will use a lot.

closed 2017-10-19 by CobyKabili