Robo3D/User-Manuals #94: R2/C2 removal of extruder and rods from grantry

CobyKabili, 2017-09-14:
People who need to remove their extruder assembly dont have instructions on how to do it. They think they may need to remove the entire gantry system. Lets create an article where we show them how to undo the 6mm rods and remove the extruder assembly that way.

mnacinopa, 2017-09-18:
@CobyKabili  Is this just to remove the extruder assembly? Or do i need to make a guide on how to change the bearings within it?

mnacinopa, 2017-09-19:
[C2](https://help.robo3d.com/hc/en-us/articles/115001624591-Servicing-Extruder-Assembly-)
[R2](https://help.robo3d.com/hc/en-us/articles/115001633792-Servicing-Extruder-Assembly)

There's no difference between the two besides a note saying C2 and R2 have similar extruder assemblies. @peries  Can you take a look at the copy whenever you can? Thanks

CobyKabili, 2017-09-20:
@mnacinopa these look good. In the R2- we have to mention that there is a second cable for the heating of Ext 2. It should be marked with a black mark i believe. We need to tell them about this.

closed 2017-10-05 by CobyKabili