Robo3D/User-Manuals #96: Electronics mainboard images and plug locations for R2/C2 and R1+

CobyKabili, 2017-10-05:
@mnacinopa we need a schematic of the electronics boards for when customers are replacing and plugging in. Something like the 3 examples below, but we need it for both the all 3 of our printers

http://reprap.org/mediawiki/images/thumb/6/6d/Rampswire14.svg/500px-Rampswire14.svg.png

http://www.hobbyist.co.nz/sites/default/files/ramps/schematic.png

http://www.achatzmediaserver.com/support/wp-content/uploads/2015/11/ramps-14-instructions-connections-3d-printer.png

mnacinopa, 2017-10-18:
@AllenMcAfee  We want to do something like what Coby linked to above.

mnacinopa, 2017-10-25:
[Mainboard Schematic](https://help.robo3d.com/hc/en-us/articles/115002394151)

mnacinopa, 2017-10-30:
@CobyKabili  
Here's the first draft of it. I asked @peries to try to see if she could clean it up a bit. 
[image]

mnacinopa, 2017-10-30:
Doubled the thickness of the wiring to make it easier to see
**R2**
[image]

**C2**
[image]

CobyKabili, 2017-10-31:
@mnacinopa these are great. One question, is there a reason the board is upside down?

AllenMcAfee, 2017-10-31:
@mnacinopa the "B" is flipped in this version as well.  Who the hell is ropo?

mnacinopa, 2017-10-31:
@CobyKabili  Not really. I made the board first then made the logo after. 

@AllenMcAfee  Some new printer company thats trying to steal our logo. 

Wiring Edits Peri made.  Changed ropo to robo, for copyright reasons. 

C2
[image]

R2
[image]

CobyKabili, 2017-11-02:
@mnacinopa we need an R1+ diagram as well. Our ramps boards are a little different than the stock off the shelf ones

mnacinopa, 2017-11-03:
[image]

CobyKabili, 2017-11-03:
looks good. Put it in an article and I can update a macro for it.

mnacinopa, 2017-11-03:
@CobyKabili 
[R1+ RAMPS Board Wiring Diagram](https://robo3d.zendesk.com/hc/en-us/articles/115002569971)

CobyKabili, 2017-11-03:
Perfect. Thank you.

closed 2017-11-03 by CobyKabili