Robo3D/User-Manuals #101: C2/R2 - How to fix creaking noise with washer

CobyKabili, 2017-10-17:
1. Loosen pulleys
2. put spacer in between pulley and bearing (pulley stl is with konrad, I also have a version we can try)
3. Tighten pulleys

ROBOLabs, 2017-10-17:
I tried this yesterday on an R2 with the STL I created, and it seems to work as expected. I will run a series of prints on the modified machine(s) and monitor the performance.

mnacinopa, 2017-10-17:
@ROBOLabs  Can you send me the stl?

ROBOLabs, 2017-10-17:
[R2 1mm PS C-clip.zip](https://github.com/Robo3D/support.robo3d.com/files/1392246/R2.1mm.PS.C-clip.zip)
I recommend printing at least 6 to a bed using a raft since they are so small.

CobyKabili, 2017-10-19:
I put 6 of them on a plate so they can print out nicely and the user wont get frustrated with it not printing. Also, I will make c2 and r2 gcode versions of these for best print results.

[R2 1mm Rod C-clip.zip](https://github.com/Robo3D/support.robo3d.com/files/1396759/R2.1mm.Rod.C-clip.zip)

mnacinopa, 2017-10-27:
[Gantry Click/Creak](https://robo3d.zendesk.com/knowledge/articles/115002455272/en-us?brand_id=114094254812) 
@ROBOLabs @CobyKabili  Can one of you upload the clip files somewhere so we can link it in the article?

closed 2017-10-27 by CobyKabili