Robo3D/User-Manuals #106: support website changes from 10/19/17

CobyKabili, 2017-10-31:
- [ ]  Make support menu item non-dropdown - links to help center site

- [ ] Make the community page on the help center link to the user forums

- [ ] Registration is within the help center

- [ ] Products, explore, support top level menu items

- [ ] Delete help@robo3d.com email from contact us. Use forms only to start tickets

- [ ] Get rid of press room on contact page from contact us

- [ ] Change general questions email

- [ ] Delete investor relations email from contact us page

- [ ] Customer support not easily accessible from contact us page. Change "Schedule an Appointment" to "Customer Support"

- [ ] Help page "Recent activity" is not obvious what it means. Change 'Recent activity" to "Trending" and add product category label

- [ ] Help page menu should have more options. Add "Submit Ticket" button, must be logged in, add "Log in to submit ticket" if not logged in

- [ ] Registering printer is only at bottom of page. Add "Register Printer" to menu

- [ ] Cura for robo desktop icon is wrong. Change icon

- [ ] "Getting Started Links" lead outside of help site. figure out a better way

- [ ] "See more" links under product help lists are hard to recognize. Make links more obvious

- [ ] No link on pages to help center home. Make a new link to bring you back to the main page

peries, 2017-12-06:
hey @bkabili how do articles get into the trending section? is it by views? also when they're listed, can you have it be like

**Article title (linked)**
Product > Category (not linked)

Right now it's 

**Category (linked)** 
Article title (linked)

peries, 2017-12-06:
Also @CobyKabili I think we talked about adding a box underneath community that links out to blog posts? Or some kind of content to show people what else they can do with their prints
[image]

bkabili, 2017-12-07:
@peries it will show the 5 most recently updated articles or topics visible to the user.

I dont believe zendesk gives access to change the layout inside of the trending section but ill have to do some more research on it.

LMK what the best thing would be for under the community section and I can add it in.

bkabili, 2017-12-07:
@peries Ive added a blog section underneath the community section. Please take a look.
https://help.robo3d.com

peries, 2017-12-07:
Cool that looks good.

Were you able to find anything on formatting the trending articles? For reference, the way that search lists the returned results would work

[image]

bkabili, 2017-12-11:
@peries @CobyKabili 
Ive looked through the zendesk support and am not seeing that there is a way to change the way results are returned for {{recent_activity}} which is what we are using in that spot.

What I can do for now is format the results with CSS so that the category is not displayed, and we only see the linked title like this:
[image]
instead of this:
[image]

It may be possible to remove the {{recent_activity}} results and instead display results from a search query to get the different "search formating" that Coby mentioned above but ill have to dig deeper to figure out how to do that with Zendesk.

For now do you want me to make the css changes above or nah?

peries, 2017-12-11:
I like it with the CSS changes. I think it's cleaner and the article title is more relevant than the category. What do you think @CobyKabili ?