Robo3D/User-Manuals #107: Changing start gcode via web dashboard for different starting temps

CobyKabili, 2017-11-02:
We have had a lot of customers call in about there temps not being set to the right temperature, even though this is just the starting priming line. We need a tutorial on how to go into the web dashboard and change the start gcode temps, and possibly other items in there as well.

mnacinopa, 2017-11-03:
@peries  Can you scrub through this? Also need to think of a better title/where to place this article
[Changing the Start temp](https://help.robo3d.com/hc/en-us/articles/115002558532)

peries, 2017-11-03:
@mnacinopa edited

mnacinopa, 2017-11-03:
@CobyKabili 
[R1+ RAMPS Board Wiring Diagram](https://robo3d.zendesk.com/hc/en-us/articles/115002569971)

peries, 2017-11-03:
should we also add it as an attachment so people can download it to see it bigger?

mnacinopa, 2017-11-03:
Thats a good idea. I'll do it.

mnacinopa, 2017-11-03:
@peries  Should be updated.

peries, 2017-11-03:
cool.

I just noticed it looks like there's an extra line going over the z motor connector on the board
[image]

mnacinopa, 2017-11-03:
@peries  Fixed.

CobyKabili, 2017-11-07:
https://help.robo3d.com/hc/en-us/articles/115002558532-Changing-the-starting-printing-temperature-via-the-start-gcode

closed 2017-11-07 by CobyKabili