Robo3D/User-Manuals #113: R2/C2 Gifs of lubricating the axis

CobyKabili, 2017-11-07:
We had a comment on this article that requested gifs or pictures to show where to lubricate the axis.

https://help.robo3d.com/hc/en-us/articles/115000715951-Lubricating-the-axis

mnacinopa, 2017-11-10:
https://help.robo3d.com/hc/en-us/articles/115000722332

@peries @CobyKabili  Please review it

peries, 2017-11-11:
@mnacinopa edited

CobyKabili, 2017-11-14:
We should have a picture or saying using the lubricant that came in the toolbox with the printer for the z axis leadscrew

mnacinopa, 2017-11-14:
@CobyKabili 
added that part.

Also, I forgot to mention to lubricate the Z Shafts (Linear bearings). For those we want to use the 3 in 1 oil like we do for the extruder bearings right? (As opposed to the bushings

peries, 2017-11-14:
Can we use the included lube for all of the parts?

mnacinopa, 2017-11-14:
We can, but according to Allen/Coby its not advised? It could gunk up the ball bearings. And something about the bushings absorbing it is also bad.

peries, 2017-11-14:
So you can use the included one for the z-axis only?