Robo3D/User-Manuals #118: Troubleshooting Guide for R1 Plus thermistor

spoogs2003, 2017-11-08:
Can we please have a troubleshooting guide for low temp error or no temp on the thermistor on the R1 plus. The guide should show how to check for connections issues on the hotend, thermistor long wire, and at the ramps board on the T0 pin.

mnacinopa, 2017-11-09:
@spoogs2003 can you send a picture of the printer spitting out a low temp error?

spoogs2003, 2017-11-09:
Hi,

I don't have a picture handing. The only visual that will prompted to say
the error would be on Mater control. A lot of the time we also determine
the issue by listening to the customers description of how the printer is
acting and then we come to that conclusion.

regards,

*Roberto Leal **B.A. History *
TECH SUPPORT

Phone: *858-213-5224*

5070 Santa Fe St., Suite C
San Diego, CA 92109

www.robo3d.com

*New Robo C2 and R2 smart 3D printers available for pre-order!*
<http://www.robo3d.com/robor2>

This email transmission contains information that is intended to
be confidential and privileged. If you receive this email and you are not a
named addressee you are hereby notified that you are not authorized to
read, print, retain, copy, or disseminate this communication without
consent of the sender and that doing so is prohibited and may be unlawful.
Please notify sender immediately by email if you have received this via
email and delete and otherwise erase the email and any copies or
attachments from your computer system. Also, email transmission cannot be
guaranteed to be secure or error free as information could be
intercepted, corrupted, lost, destroyed, arrive late or incomplete, or
contain viruses. The sender therefore does not accept liability for
any errors or omissions in the contents of the message, which arise as a
result of email transmission.