Robo3D/User-Manuals #121: Article Organization and symptom explaination

CobyKabili, 2017-11-21:
We received good feedback from a customer with a simple leveling issue, but could have been solved by himself. He brings up good points and I would like to figure out a way to solve the issue of not being able to search for a problem that you are having. I was thinking of an article or section that could be titled what the symptom is, and then linking to the articles to try and solve this system. Lets brainstorm. Email response from customer below to give context.

_"Hi Dave,Thanks for following up.Yes, your solution did work.  Close the ticket.However, then I ran into other problems, mostly with adhesion.  I have used empirical methods (trial and error), and read everything I could find on the problem and seem  to have worked that one out.My first successful print is printing now.  I hope it completes.If you will allow me to make a suggestion, and it will be essentially the same on my Amazon review.  Your machine fills a very important niche in a growing 3D printer market: namely, the first printer for "newbies" such as myself.  There is a significant potential barrier against adopting 3d printers because the difficulty appears overwhelming.  (Maybe not to you because you live this stuff.)  The Robo R1+ seems to fit right in as the first printer to learn the technology.  It is mechanically robust, the software is excellent, and the design is easy to use.  Frankly, your support is pretty good too.However, your documentation is atrocious.  I know you (Robo) and most software / high tech companies love the Internet, and now all help is WEB based.  It is awful.  You can never find what you want because it is not logically organized; the assumption is to allow the search function to find out what you want.  But, that it problematic at best, and is almost always out of context (think Microsoft Office help).  Indeed, some of the articles make no sense anyway.  E.G. "Misaligned R1+ Z-axis.  This article shows how to align the Z-axis; I think.  But the Robo R1 does automatic leveling of the Z-axis.  Why would I want to do it manually???Also, most of your articles are titled by the method of repair, not the symptom.  So, it is not obvious which articles might apply to which problems.  Indeed, in your letter to me you pointed out the relevant articles, but I would never have found them by myself.So, here is a suggestion:Put together decent documentation, like a users manual.  In the manual list all of the problems that a new user is likely to encounter.  I.e. choose those problems that cover 90% of the problems most people have.  Then, make a list and provide a nice explanation for each one that explains why the problem occurs and what to do about it.For example: Extruder extrudes fine except when in contact with print bed."  Then explain the offset.  But, also add that if you make the offset too high, you will get adhesion problems. Have a list of these.  Then the poor newbie user like me can scroll down, find his problem, and understand it right away.Kindest Regards"_
--

peries, 2017-11-21:
Yeah I hear what he is saying. Michael and I have had discussions on this as well. Michael was saying if we title them based on the symptom, there might be a multitude of problems/solutions related to that symptom which would make the article really long and possibly overwhelming to the reader. 

But yeah let's brainstorm. I'm sure we can come up with a better way to break it down

mnacinopa, 2017-11-21:
@CobyKabili 's suggestion is that we have an article with the symptom, and just include all the links to the possible solution within that article.

CobyKabili, 2017-11-28:
Here is what mosaic does with palette+. I like it a lot. Check it out

https://mosaicmanufacturing.zendesk.com/hc/en-us/articles/115001512527-Error-Numbers

mnacinopa, 2017-12-01:
@CobyKabili @peries @ROBOLabs 
Can you guys help me come up with a list of symptons for all the machines? I'm going to be creating a Master Troubleshooting  article (one for each machine) that will have Symptoms and then link to and explain helpful/relevant articles. 

I have a draft right here. Going to put it at the top of each troubleshooting section. 
[Troubleshooting Index](https://help.robo3d.com/hc/en-us/articles/115003103092)

mnacinopa, 2017-12-22:
First draft up for the[ R2/](https://help.robo3d.com/hc/en-us/articles/115003103092) [R1+ ](https://help.robo3d.com/hc/en-us/articles/115003584291)

@peries 
@AllenMcAfee 
@CobyKabili